ckcr4lyf/EvilAppleJuice-ESP32 #2: Not spammy when using ESP32-WROOM-32

ExME168, 2023-09-16:
I've tested this on several iOS 16 devices (one is using iOS 16.2 and the others are using the latest iOS 16.6.1) using an ESP32-WROOM-32 and it does not produce the same spamming effect seen in your demo video. I've reviewed the datasheets of the ESP32-WROOM-32 and the ESP32-C3 and it turns out the former only supports up to BLE 4.2 whereas the latter can support up to BLE 5 (and a bunch more features like simultaneous advertising and scanning and advertising extensions). It's very likely that the ESP32-WROOM-32 is encountering hardware limitations and that's why I can't spam the test devices. 

So yeah, if anyone is going to use this tool, make sure to either use the same ESP32-C3 used by the developer or at least any microcontroller with BLE 5 support.

ESP32-WROOM-32 Bluetooth Specs:
[image]

ESP32-C3 Bluetooth Specs:
[image]
[image]

ExME168, 2023-09-16:
[image]
Yeah it's able to set the TxAdd parameter to random

ckcr4lyf, 2023-09-16:
That's odd, if it is the same BLE packet I would imagine it doesn't matter what the source device is. What iPhone / iOS are you testing it with?

Unfortunately I only have ESP32-C3s from AirM2M with me, but I can try and target the same iOS device and see the results.

ExME168, 2023-09-16:
The only iPhone I have on-hand right now is an iPhone X running iOS 14.8. I'm pretty sure this is way more stubborn compared to iOS 16 devices because the only notification I keep seeing from this is the spoofed "password autofill" for Apple TV (that is index 6 in the short devices array). Sometimes it pops up other Apple TV notifs like Apple TV setup and color balance but it's rare. It never pops up notifs related to audio devices.

As for the iOS 16 devices,
I don't know what iPhone my friend was using exactly (most likely iPhone 13 or 14 because it had three lenses) but it was surely running iOS 16.2. His phone got the popup notifs way more frequently but still I wouldn't consider it "spammy". There were times when the notifs just stopped popping up. The iOS 16.6.1 devices I tested were literally iPhones displayed in a store and they just brought up the popups for the first couple of seconds but then stopped getting them. Still not the spammy effect.

ckcr4lyf, 2023-09-16:
Hmm, let me try in an Apple store as well and get back to you.

h0e4a0r1t, 2023-09-19:
I use Arduino IDE to program ESP32-WROOM-32 and it can be used normally

ckcr4lyf, 2023-09-19:
@h0e4a0r1t are you observing the "spammy" popups on an iPhone with the ESP32-WROOM-32? If so, could you share the iPhone version / iOS version as well

h0e4a0r1t, 2023-09-19:
Yes, I can get "spam" popping up on my iPhone. My phone model is iPhone 13/iOS version is 16.6.1

cryptr0n1x, 2023-09-20:
Testing with ESP32-WROOM-32 and iPhone 13 / iOS 16.6. Unfortunately not "spammy" enough.

cryptr0n1x, 2023-09-20:
> Yes, I can get "spam" popping up on my iPhone. My phone model is iPhone 13/iOS version is 16.6.1

Could you show us a video, how "spammy" it is?

I tried with RPi + BT Dongle and ESP32-WROOM-32. Both versions didn't get any spam effect on my iPhone 13 mini.

After closing one or two popups nothing happend. Spammy means to me, you cannot use your iPhone anymore XD

ckcr4lyf, 2023-10-11:
> Can I use any ESP32 circuit or does it have to be an ESP32-C3?

Which ESP32 is it? If it has a BLE radio it could work, but I've not tested it.

ckcr4lyf, 2023-10-13:
> Module Wifi BLE ESP32 Node MCU LuaNode32 CP2102

Looking at some of the boards w/ a generic google search, seems it is a ESP32-WROOM-32. From what others have mentioned in the thread, it may not work as well as an ESP32-C3.

If you live in Asia, and have some kind of Taobao forwarding service available, ESP32C3 on their is only like 13RMB, would recommend it. (And if you get multiple, the effect is even better).

louis16s, 2023-10-13:
ESP32C3 is only 9.9RMB now in China

louis16s, 2023-10-13:
u should rename it into esp32-c3 cause esp32 is something different

ckcr4lyf, 2023-10-13:
> u should rename it into esp32-c3 cause esp32 is something different

Technically it should work on any ESP32 with a BLE stack. I've only tested on the C3 because that's all I have.

ckcr4lyf, 2023-10-13:
> however i think it not "spammy" enough

Which iPhone / iOS? And what distance?

AgonyListener, 2023-10-13:
> > however i think it not "spammy" enough
> 
> Which iPhone / iOS? And what distance?

13pro17.0.2 and ipadpro m2 ios16.6.1

MrTechGadget, 2023-10-13:
Spammy enough with my wemos_d1_mini32 (which uses an ESP32-WROOM-32) when within one foot of the iPhone. Works less often from 2 feet away. So I suspect it is just a power output issue.

xIvan0ff, 2023-10-14:
Same here, iPhone 12 Pro iOS 17.0.3 with ESP32-WROOM-32, spammy when literally touching the phone, but if it's like 20-30 cm away I am only getting the Autofill notification.

reecepounder, 2023-10-28:
> Same here, iPhone 12 Pro iOS 17.0.3 with ESP32-WROOM-32, spammy when literally touching the phone, but if it's like 20-30 cm away I am only getting the Autofill notification.

This behavior might be attributed to adjustments by Apple. I remember reading somewhere that they were altering the required RX power detected in order to present these notifications - it would make sense that the Apple TV one is exempt as it is unlikely to be used in close proximity to the device whereas new headphones / other devices would generally be first used right next to device. I have no evidence to support this but am doing some of my own testing with ESP-WROOM-32.

ckcr4lyf, 2023-10-28:
>  It would make sense that the Apple TV one is exempt as it is unlikely to be used in close proximity to the device whereas new headphones / other devices would generally be first used right next to device.

Very interesting hypothesis. In my random testing at apple stores, I've also observed that the Apple tv popup is the most "easily spread", i.e. over distances or 4-5+ metres, all iphones pop it up pretty quickly. 

I will try this out as well and see if it gives better results "at a distance".

Houraisan-yume, 2023-11-07:
I’ve met the same question, the board is Esp32c3 brought from LUATOS TaoBao store.
I tested 3 devices ipad9, iphone14pm, iphone13, all of them running ios 17.1.
At the first time flashing and running, everything works fine and spammy as expected.
But after reconnect the wire on the esp32, the iphone seems only show apple tv related messages and the ipad only shows the earphone related. 
Change distance does not help, and the apple tv messages have a longer distance to be received, but obviously lower frequency than the earphone messages.
I tried re-flashing or reset the board but still the same perform. 
Just feel weird cause the first running was normal, and the advertising behavior of the board seems normal as well.
I guess may be iphone and ipad can selectivity ignore some of these messages.

ckcr4lyf, 2023-11-07:
\> But after reconnect the wire on the esp32

You mean just take off power and turn it back on?

Anecdotally, in Apple Store I have seen most success with Apple TV messages. Maybe I can add an option to prioritize this guy.

Houraisan-yume, 2023-11-07:
yep, just take it off and replug the usbc, then couldn't function spamming the iphone devices except apple tv related. 
I've tried some steps to reset or reflash but still only the first time seems useful.

MagicBlack1017, 2023-11-07:
> I’ve met the same question, the board is Esp32c3 brought from LUATOS TaoBao store. I tested 3 devices ipad9, iphone14pm, iphone13, all of them running ios 17.1. At the first time flashing and [...]

Me too, my board model is ESP-WROOM-32. I have tested it on multiple Apple devices, but every time it only pops up the Apple TV notification, and only once.  It only reappears after locking and unlocking the screen again.

mark304188, 2023-11-13:
I tried it with my esp32-wroom on  some devices (iPhone and iPad),and it would only get Apple TV related popups,even the distance is closed enough . When it's far enough like 30cm or so,it won't show anything .

Ayndpa, 2023-11-13:
It works on my ESP-WROOM-32 but it's far from being spammy. Only AirPods trigger notifications, and only once.I didn't even see Apple TV notifications.

Ayndpa, 2023-11-13:
> It works on my ESP-WROOM-32 but it's far from being spammy. Only AirPods trigger notifications, and only once.I didn't even see Apple TV notifications.

Correction: It only triggers when the screen is on or off, and only with AirPods.

Ayndpa, 2023-11-13:
This also occurred when I previously tried using the Android app "nRF Connect." The same issue was present.

ckcr4lyf, 2023-11-15:
I've just tried my existing ESP32-C3's, and on one iPhone w/ iOS 17.0.3 it is only popping up Apple TV / Migrate Phone number etc. notifications, and just once for each.

A lock+unlock of the screen will allow them to come up again.

Did not see any for Airpods / Beats. I will test a few more iPhones tonight, but seems Apple may have patched it a bit (at least reduced the frequency).

ckcr4lyf, 2023-11-23:
FYI: A user confirmed it working on iPhone 14 Pro Max (running iOS 17.2 b3)

See: #19

JK-2292, 2023-11-29:
> I've just tried my existing ESP32-C3's, and on one iPhone w/ iOS 17.0.3 it is only popping up Apple TV / Migrate Phone number etc. notifications, and just once for each.
>
> A lock+unlock of the screen [...]

 Was working on ESP cores & stumbled across your project here.  In terms of "real world" application, I believe you all are looking 4-this.....  ESP32-C3-WROOM-02U Dev board.  Find it, closely read the datasheet 4-this Dev board.  The C6 series is better but the optimal configuration isn't offered in  NA yet.

tjs20630, 2023-11-29:
I'm using iPhone 11 with iOS 16.5 alongside the ESP-WROOM-32, and I'm experiencing a less aggressive spamming of advertisements compared to what was demonstrated in the video. It raises the question of whether this discrepancy could be attributed to the specific ESP32 board I'm using, as previously mentioned. Interestingly, Apple TV windows briefly appeared a few times, but there were no subsequent interactions, not even with Airpods. I'm curious if this is solely a board-related issue or if there are other potential factors contributing to this behavior.

itsrealkk, 2024-01-15:
I'm using iPhone 15 Pro Max with IOS 17.2.1. My dev board is also ESP32-WROOM-32, I can use Arduino IDE to upload to the dev board. It can successfully run on my phone.

FutzingKlutz, 2026-02-11:
Just tested this, had to remove ArduinoBLE and just use ESP32's bluetooth library. Then that broke other things, fixed some of them. Looks almost okay in serial monitor.

`#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLEAdvertising.h>
#include "devices.hpp"

BLEAdvertising *pAdvertising;
uint32_t delayMilliseconds = 100; // Delay between advertisements

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE Advertising...");

  // Initialize BLE
  BLEDevice::init("AirPods 69");

  // Create BLE Server
  BLEServer *pServer = BLEDevice::createServer();

  // Get advertising instance
  pAdvertising = BLEDevice::getAdvertising();

  Serial.println("BLE Initialized. Starting advertisements...");
}

void loop() {
  // Calculate array sizes correctly
  int numLongDevices = sizeof(DEVICES) / sizeof(DEVICES);
  int numShortDevices = sizeof(SHORT_DEVICES) / sizeof(SHORT_DEVICES);

  // Advertise long devices
  for (int i = 0; i < numLongDevices; i++) {
    advertiseLongDevice(DEVICES[i]);
    delay(delayMilliseconds);
  }

  // Advertise short devices
  for (int i = 0; i < numShortDevices; i++) {
    advertiseShortDevice(SHORT_DEVICES[i]);
    delay(delayMilliseconds);
  }
}

void advertiseLongDevice(const uint8_t* data) {
  BLEAdvertisementData oAdvertisementData = BLEAdvertisementData();
  
  // Set flags (0x1a = LE General Discoverable Mode + BR/EDR Not Supported)
  uint8_t flags = 0x1a;
  oAdvertisementData.addData((char*)&flags, 1);
  
  // Set manufacturer data (31 bytes)
  oAdvertisementData.addData((char*)data, 31);
  
  // Apply advertisement data and start
  pAdvertising->setAdvertisementData(oAdvertisementData);
  pAdvertising->start();
  
  Serial.print("Advertising long device: ");
  for (int i = 0; i < 31; i++) {
    Serial.printf("%02X ", data[i]);`