ckcr4lyf/EvilAppleJuice-ESP32 #3: Don't working on ESP32-WROOM-32

Clinner0design, 2023-09-17:
[image]
Always like this. The port monitor says the same thing

ckcr4lyf, 2023-09-18:
@Clinner0design the log is expected, the serial won't print anything else. Was there some other behavior you were expecting?

Maybe I can add more log statements to make the initialization clear.

When you say "not working" - do you mean that there is no notification on an iPhone? Do you have another device capable of capturing BLE traffic (e.g. Android with nRF connect app), and can you see if packets are being sent out?

ckcr4lyf, 2023-09-18:
FYI: another user did report actual issues with the WROOM-32 in terms of the "spammy" nature

See: https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/issues/2

ahmad779900, 2023-09-19:
i can confirm too it's not spammy as it looks in the video

ckcr4lyf, 2023-09-19:
@ahmad779900 which ESP, which iPhone/iOS?

ahmad779900, 2023-09-19:
esp wroom 32 with my friend's iphone xs max

ckcr4lyf, 2023-09-19:
it seems the iPhone X is more stubborn, it doesn't show the airpods/beats notifications. Can you try and modify the script to only spam the AppleTVs and see what happens?

The easiest way if you're not too proficient with C++ would be to change this line:

https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/blob/336aa4143930117bf3fc684323d844694123b78e/src/EvilAppleJuice-ESP32-INO/EvilAppleJuice-ESP32-INO.ino#L82

to `int device_choice = 1;`

and see how it goes. I'll add some more logging today as well.

ckcr4lyf, 2023-10-11:
Closing as dupe of https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/issues/2 , we will discuss ESP32-WROOM-32 in there.