ckcr4lyf/EvilAppleJuice-ESP32 #10: Led lights have been added to indicate the status.

ahgitit, 2023-10-29:
[EvilAppleJuice-ESP32-master_led_flash.zip](https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/files/13198102/EvilAppleJuice-ESP32-master_led_flash.zip)

ckcr4lyf, 2023-10-29:
Sorry, can you explain what this issue is about? I don't want to download an arbitrary zip file.

ahgitit, 2023-10-29:
> Sorry, can you explain what this issue is about? I don't want to download an arbitrary zip file.

I made a new change, so that every time esp32 sends a signal, the leds will flash once.This allows us to intuitively know that he has carried out an attack.

ckcr4lyf, 2023-10-30:
I can't do anything with just a zip file. Additionally I don't believe this adds to much value to the current scope of the project, since there is not guarantee that this had any effect on any iOS device.

You can surely use it in your own fork, and maybe link it here if you think it may help someone.

ckcr4lyf, 2023-11-06:
Closing as no further comment.

closed 2023-11-06 by ckcr4lyf