ckcr4lyf/EvilAppleJuice-ESP32 #11: A dumb question about delaySeconds

sht2017, 2023-11-01:
Hi,
I have no experience with esp32 and its Bluetooth library, Bluetooth protocal, etc. before. I was just wondering if the delaySeconds is smaller, the frequency of spam advertisements would be higher? Or limited to its hardware, could it be too short to send advertisements or there is a built-in filter somehow to filter it on iOS?

ckcr4lyf, 2023-11-01:
Not a dumb question at all. I am guessing you're talking about this guy:

https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/blob/8effcdac60f8538954c2c68a2cd8534312ee43a0/src/main.cpp#L13

I do believe, at some level, it would be limited by the underlying ESP32 hardware. It may also be somewhat limited on the iDevice side, e.g. if it sees "too many" advertisements in a very short interval, it would think they are fake.

Unfortunately, I do not have an iDevice personally, so I am unable to perform fine grained testing on the delay parameter to see how it affects the spamming.

You could definitely try and play around with it, to see if you can spam it even more frequently (since also in every loop it will randomize the spoofed device)

sht2017, 2023-11-01:
Thanks for your response, I'll look into it =)

ckcr4lyf, 2023-11-01:
No problem. Btw. if you're unfamiliar with C, you can't actually have a fractional value (E.g. 0.5) for a `uint32_t` , but you could change the amount is is multiplied by, since the value passed to the actual delay function is in milliseconds:

https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/blob/8effcdac60f8538954c2c68a2cd8534312ee43a0/src/main.cpp#L112

ckcr4lyf, 2023-11-06:
@sht2017 it may interest you, that in the official Apple Design Specification (https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf), they mention, quote: "The accessory should first use the recommended advertising interval of 20 ms for at least 30 seconds."

Seems like every 20ms (which means ~50 times a second!) is within Apple spec!

[image]

I will do some PoC, using the current 1 second sleep, and a much shorter / reduced 50ms sleep, and report back if I see any noticeable changes.

Reopening this issue so others may also come across it and try it out

sht2017, 2023-11-06:
> @sht2017 it may interest you, that in the official Apple Design Specification (https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf), they mention, quote: "The accessory should [...]

Thanks for the info, I'll check that. 
Also, I noticed that trying it at my local Apple Store didn't work as well as before (I'm currently using ESP32-S3 with `delaySeconds = 200`). Only devices really nearby would perform inactive popups. So I assume Apple may already applied an update with an aggressive detection filter algorithm to filter devices with weak BT signals.

sht2017, 2023-11-06:
Btw, I guess we may talk about this further within a new issue so that could keep everyone else on update.

closed 2023-11-14 by sht2017