ckcr4lyf/EvilAppleJuice-ESP32 #12: Help for download issues

King7749, 2023-11-06:
Seeking help, I saw in your project video that you used Esp32-C3 from Hezhou. However, after I successfully burned it through Arduino, the serial port did not output. The Esp32 library confirmed the correct addition, but the serial port did not output any content (with a baud rate of 115200). May I ask how you burned it

ckcr4lyf, 2023-11-06:
I used PlatformIO and Visual Studio Code. I'm guessing you tried the Arduino IDE? Can you try PlatformIO once?

ckcr4lyf, 2023-11-06:
Make sure this is set in your PlatformIO config if you're not cloning the repo:

https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/blob/8effcdac60f8538954c2c68a2cd8534312ee43a0/platformio.ini#L11-L15

King7749, 2023-11-06:
> 我使用了 PlatformIO 和 Visual Studio Code。我猜你试过Arduino IDE？您可以尝试一次 PlatformIO 吗？

Yes, I tried the Arduino IDE, but unfortunately it didn't work, which drove me crazy. I haven't used PlatformIO before, but I will follow your advice and give it a try. Thank you very much for your help

ckcr4lyf, 2023-11-06:
I'll also try and make a video tutorial in the future for new users.

King7749, 2023-11-06:
> 我将来还会尝试为新用户制作视频教程。

I'm looking forward to it. Thank you

King7749, 2023-11-06:
> > > 我将来还会尝试为新用户制作视频教程。
> > 
> > 
> > 我很期待。谢谢
> 
> 存储库的核心不使用任何特定于硬件的代码（在 esp32 的某些型号或变体上），所以我想你最终可能会找到如何编译和运行项目。如果您遇到任何问题，请告诉我，我很乐意为您提供帮助:)

Thank you, I have solved it. It seems that the Esp32-C3 of Hezhou is not friendly to the Arduino IDE download method, but I successfully downloaded it using PlatformI and it has been effective. I would like to express my gratitude to ckcr4lyf for his help and your kind help. Thank you very much!!!

huang1008686, 2023-11-14:
寻求帮助，您好，我是初学者，手上的板子是ESP32-S3-WROOM-1(N16R8),请问在VSCode+platformIO新建选择板子应该选哪个，没有我说的这个板子可以选，工程文件里面的platform.ino需要修改吗，试了好多次无法成功烧录，希望能得到一丝帮助，十分感谢

ckcr4lyf, 2023-11-14:
@huang1008686 

Maybe this can be of help:
* https://community.platformio.org/t/cant-find-esp32-s3-wroom-1-n16r8-16m-flash-8sram/30258
* https://github.com/platformio/platform-espressif32/pull/921#issuecomment-1766069497

I don't have the exact board so I cannot be sure.

sht2017, 2023-11-14:
> 寻求帮助，您好，我是初学者，手上的板子是ESP32-S3-WROOM-1(N16R8),请问在VSCode+platformIO新建选择板子应该选哪个，没有我说的这个板子可以选，工程文件里面的platform.ino需要修改吗，试了好多次无法成功烧录，希望能得到一丝帮助，十分感谢

Try it 
>platform.ini
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200

```

King7749, 2023-11-15:
> > 寻求帮助，您好，我是初学者，手上的板子是ESP32-S3-WROOM-1（N16R8），请问在VSCode+platformIO新建选择板子应该选哪个，没有我说的这个板子可以选，工程文件里面的platform.ino需要修改吗，试了好多次无法成功烧录，希望能得到一丝帮助，十分感谢
> 
> 尝试一下
> 
> > 平台.ini
> 
> ```
> ; PlatformIO Project Configuration File
> ;
> ;   Build options: build flags, source filter
> ;   Upload options: custom upload port, speed and extra flags
> ;   Library options: dependencies, extra library storages
> ;   Advanced options: extra scripting
> ;
> ; Please visit documentation for the other options and examples
> ; https://docs.platformio.org/page/projectconf.html
> 
> [env:esp32-s3-devkitc-1]
> platform = espressif32
> board = esp32-s3-devkitc-1
> framework = arduino
> monitor_speed = 115200
> ```

我觉得你可以试试通过arduion来进行烧录，因为它具有esp32-S3的库

sht2017, 2023-11-15:
> 我觉得你可以试试通过arduion来进行烧录，因为它具有esp32-S3的库

I think so.
But considering uncertainty, using Platformio might still be the best solution here. 
Btw, I am using esp32-s3 with it and it's fine once you apply the previous config to platform.ini.

huang1008686, 2023-11-15:
> > > > 寻求帮助，您好，我是初学者，手上的板子是ESP32-S3-WROOM-1（N16R8），请问在VSCode+platformIO新建选择板子应该选哪个，没有我说的这个板子可以选，工程文件里面的platform.ino需要修改吗，试了好多次无法成功烧录，希望能得到一丝帮助，十分感谢
> > > 
> > > 
> > > 尝试一下
> > > > 平台.ini
> > > 
> > > 
> > > ```
> > > ; PlatformIO Project Configuration File
> > > ;
> > > ;   Build options: build flags, source filter
> > > ;   Upload options: custom upload port, speed and extra flags
> > > ;   Library options: dependencies, extra library storages
> > > ;   Advanced options: extra scripting
> > > ;
> > > ; Please visit documentation for the other options and examples
> > > ; https://docs.platformio.org/page/projectconf.html
> > > 
> > > [env:esp32-s3-devkitc-1]
> > > platform = espressif32
> > > board = esp32-s3-devkitc-1
> > > framework = arduino
> > > monitor_speed = 115200
> > > ```
> > 
> > 
> > 我觉得你可以试试通过arduion来进行烧录，因为它具有esp32-S3的库
> 
> 我认为如此。但考虑到不确定性，使用 Platformio 可能仍然是最好的解决方案。顺便说一句，我正在使用 esp32-s3，一旦您将之前的配置应用于平台 .ini，它就很好了。
> 
> 此致敬意

非常感谢您的回复
昨晚在作者发的论坛中给了我很大的思路，板子选择这款可以在platformio中烧录了，加上前辈您发了修改的ino文件，成功烧录，但貌似发射距离不能超过一米，不知道是不是功率还是延时问题，变慢了
此致敬意

huang1008686, 2023-11-15:
> 我觉得你可以试试通过arduion来进行烧录，因为它具有esp32-S3的库

感谢前辈的回复
用arduino烧录会出现不少问题，库的路径引用总是有问题，这次修改选对板子直接用platform烧录就成功了
[image]

huang1008686, 2023-11-15:
> 我认为如此。但考虑到不确定性，使用 Platformio 可能仍然是最好的解决方案。顺便说一句，我正在使用 esp32-s3，一旦您将之前的配置应用于平台 .ini，它就很好了。
>
> 此致敬意

非常感谢您的回复
昨晚在作者发的论坛中给了我很大的思路，板子选择这款可以在platformio中烧录了，加上前辈您发了修改的ino文件，成功烧录，但貌似发射距离不能超过一米，不知道是不是功率还是延时问题，变慢了
[image]

此致敬意

huang1008686, 2023-11-15:
> @huang1008686
> 
> 也许这会有所帮助：
> 
> * https://community.platformio.org/t/cant-find-esp32-s3-wroom-1-n16r8-16m-flash-8sram/30258
> * [新增乐鑫 ESP32-S3-DevKitC-1-N16R8V 开发板文件 platformio/platform-espressif32#921 （注释）](https://github.com/platformio/platform-espressif32/pull/921#issuecomment-1766069497)
> 
> 我没有确切的电路板，所以我不能确定。

感谢您的回复
您发的链接给了我莫大的帮助，已经成功烧录
此致敬意

sht2017, 2023-11-15:
> > > > 寻求帮助，您好，我是初学者，手上的板子是ESP32-S3-WROOM-1（N16R8），请问在VSCode+platformIO新建选择板子应该选哪个，没有我说的这个板子可以选，工程文件里面的platform.ino需要修改吗，试了好多次无法成功烧录，希望能得到一丝帮助，十分感谢
> > > 
> > > 
> > > 尝试一下
> > > > 平台.ini
> > > 
> > > 
> > > ```
> > > ; PlatformIO Project Configuration File
> > > ;
> > > ;   Build options: build flags, source filter
> > > ;   Upload options: custom upload port, speed and extra flags
> > > ;   Library options: dependencies, extra library storages
> > > ;   Advanced options: extra scripting
> > > ;
> > > ; Please visit documentation for the other options and examples
> > > ; https://docs.platformio.org/page/projectconf.html
> > > 
> > > [env:esp32-s3-devkitc-1]
> > > platform = espressif32
> > > board = esp32-s3-devkitc-1
> > > framework = arduino
> > > monitor_speed = 115200
> > > ```
> > 
> > 
> > 我觉得你可以试试通过arduion来进行烧录，因为它具有esp32-S3的库
> 
> I think so. But considering uncertainty, using Platformio might still be the best solution here. Btw, I am using esp32-s3 with it and it's fine once you apply the previous config to platform.ini.
> 
> Best Regards

Don't worry. 
About I think we have been discussing it in #2 for a while. You can use Wireshark to analyse the BLE packet. For now, I believe there already has been a patch applied on iOS which makes the frequency lower.

sht2017, 2023-11-15:
> 我觉得你可以试试通过arduion来进行烧录，因为它具有esp32-S3的库

And also, I think the issue has been solved. 
I guess you may consider closing it in order to avoid notifications if you'd like to =)

sht2017, 2023-11-15:
@huang1008686 
Don't worry.
About the frquency I think we have been discussing it in #2, #11 for a while. You can use Wireshark to analyse the BLE packet. For now, I believe there already has been a patch applied on iOS which makes the frequency lower.

silverdragon830, 2023-12-31:
i tried using esp32 devkitv1 doesnt work maybe board too old?

Xzhh6536, 2024-03-22:
帅哥可以弄一个导入教学视频吗，我是一个初学者，看不懂，好想学，呜呜呜

huang1008686, 2024-03-22:
我看看,你手上有板子吗,是指怎样烧录还是怎样,烧录的话环境配好就很好操作了

ckcr4lyf, 2024-03-22:
I'm only speaking from translate, but it seems like a video would help. I'll try and make one tomorrow. 

You would need to have an ESP32 in the first place to use it though.

King7749, 2024-03-22:
> 帅哥可以弄一个导入教学视频吗，我是一个初学者，看不懂，好想学，呜呜呜

是这样的，现在苹果貌似修复了它，弹窗频率已经大大降低了，如果你仍执意进行操作，请访问该链接图片底部有一键烧录的教程：https://whyinfo.link/posts/03-%E8%8B%B9%E6%9E%9C%E8%AE%BE%E5%A4%87%E6%97%A0%E9%99%90%E5%BC%B9%E7%AA%97（在底部三有教程）

King7749, 2024-03-22:
> 我看看,你手上有板子吗,是指怎样烧录还是怎样,烧录的话环境配好就很好操作了
> [...](#)

如上述评论参考

Xzhh6536, 2024-03-22:
> 我看看,你手上有板子吗,是指怎样烧录还是怎样,烧录的话环境配好就很好操作了
> [...](#)
> ---源语言--- 来自： ***@***.***> 日期： 周五， 2024年3月22日 21：19 收件人：***@***.***>; 抄送：***@***.******@***.***>; 主题：回复：[ckcr4lyf/EvilAppleJuice-ESP32] 下载问题帮助 （Issue#12） 帅哥可以弄一个导入教学视频吗，我是一个初学者，看不懂，好想学，呜呜呜 — 直接回复此电子邮件，在 GitHub 上查看，或取消订阅。 您收到此消息是因为有人提到了您。消息 ID：***@***.***>

我有一个esp8266开发板，我是一个小白，不知道怎么烧录

Xzhh6536, 2024-03-22:
> > 帅哥可以弄一个导入教学视频吗，我是一个初学者，看不懂，好想学，呜呜呜
> 
> 是这样的，现在苹果貌似修复了它，弹窗频率已经大大降低了，如果你仍执意进行操作，请访问该链接图片底部有一键烧录的教程：https://whyinfo.link/posts/03-%E8%8B%B9%E6%9E%9C%E8%AE%BE%E5%A4%87%E6%97%A0%E9%99%90%E5%BC%B9%E7%AA%97 （在底部三有教程）

里面的烧录文件和软件变404了

King7749, 2024-03-22:
> 我有一个esp8266开发板，我是一个小白，不知道怎么烧录

很抱歉，8266并不支持蓝牙，所以无法烧录，必需是esp32,优选esp32c3

King7749, 2024-03-22:
> > > 帅哥可以弄一个导入教学视频吗，我是一个初学者，看不懂，好想学，呜呜呜
> > 
> > 
> > 是这样的，现在苹果貌似修复了它，弹窗频率已经大大降低了，如果你仍执意进行操作，请访问该链接图片底部有一键烧录的教程：https://whyinfo.link/posts/03-%E8%8B%B9%E6%9E%9C%E8%AE%BE%E5%A4%87%E6%97%A0%E9%99%90%E5%BC%B9%E7%AA%97 （在底部三有教程）
> 
> 里面的烧录文件和软件变404了

你再试试，我这边测试可直接下载

Xzhh6536, 2024-03-22:
> 很抱歉，8266并不支持蓝牙，所以无法烧录，必需是esp32，优选esp32c3

好吧我这几天买一个试试，不过不得不说你们真的很厉害，很羡慕你们

Xzhh6536, 2024-03-22:
> > > > 帅哥可以弄一个导入教学视频吗，我是一个初学者，看不懂，好想学，呜呜呜
> > > 
> > > 
> > > 是这样的，现在苹果貌似修复了它，弹窗频率已经大大降低了，如果你仍执意进行操作，请访问该链接图片底部有一键烧录的教程：https://whyinfo.link/posts/03-%E8%8B%B9%E6%9E%9C%E8%AE%BE%E5%A4%87%E6%97%A0%E9%99%90%E5%BC%B9%E7%AA%97 （在底部三有教程）
> > 
> > 
> > 里面的烧录文件和软件变404了
> 
> 你再试试，我这边测试可直接下载

烧录工具可以下载了，不过那个安卓app还是404哦

King7749, 2024-03-22:
> > > > > 帅哥可以弄一个导入教学视频吗，我是一个初学者，看不懂，好想学，呜呜呜
> > > > 
> > > > 
> > > > 是这样的，现在苹果貌似修复了它，弹窗频率已经大大降低了，如果你仍执意进行操作，请访问该链接图片底部有一键烧录的教程：https://whyinfo.link/posts/03-%E8%8B%B9%E6%9E%9C%E8%AE%BE%E5%A4%87%E6%97%A0%E9%99%90%E5%BC%B9%E7%AA%97 （在底部三有教程）
> > > 
> > > 
> > > 里面的烧录文件和软件变404了
> > 
> > 
> > 你再试试，我这边测试可直接下载
> 
> 烧录工具可以下载了，不过那个安卓app还是404哦

再试试

King7749, 2024-03-22:
> > > > > 帅哥可以弄一个导入教学视频吗，我是一个初学者，看不懂，好想学，呜呜呜
> > > > 
> > > > 
> > > > 是这样的，现在苹果貌似修复了它，弹窗频率已经大大降低了，如果你仍执意进行操作，请访问该链接图片底部有一键烧录的教程：https://whyinfo.link/posts/03-%E8%8B%B9%E6%9E%9C%E8%AE%BE%E5%A4%87%E6%97%A0%E9%99%90%E5%BC%B9%E7%AA%97 （在底部三有教程）
> > > 
> > > 
> > > 里面的烧录文件和软件变404了
> > 
> > 
> > 你再试试，我这边测试可直接下载
> 
> 烧录工具可以下载了，不过那个安卓app还是404哦

再试试，貌似该软件已有更新版本，请在内部找到github链接下载最新版本

Xzhh6536, 2024-03-22:
> 再试试，貌似该软件已有更新版本，请在内部找到github链接下载最新版本

可以下载了，是的他有一个1.0.8新版本的，是不是还有一个BadAppleJuicek软件，作者大大

huang1008686, 2024-03-22:
> 再试试，貌似该软件已有更新版本，请在内部找到github链接下载最新版本

您这个封装的烧录工具貌似解压不了，解压密码错误，是否可以再提供一下

ckcr4lyf, 2024-03-27:
> 我有一个esp8266开发板，我是一个小白，不知道怎么烧录

FYI: I've done all my testing on an ESP32C3 from LuatOS, I would recommend it: https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/issues/12#issuecomment-2015204624

song940, 2024-03-28:
Hello 各位，我已经提交了 pull request 来自动构建预编译的固件，同时更新了 README 支持通过浏览器网页写入固件，大大降低了普通用户的使用门槛，相信大家都能轻松掌握，大家可以关注一下：https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/pull/38