ckcr4lyf/EvilAppleJuice-ESP32 #25: XIAO C3

silverdragon830, 2024-01-06:
19:13:43.583 -> load:0x403ce710,len:0x25f4
19:13:43.583 -> entry 0x403cc710
19:13:43.755 -> Starting ESP32 BLE
19:13:44.663 -> ESP-ROM:esp32c3-api1-20210207
19:13:44.663 -> Build:Feb  7 2021
19:13:44.663 -> rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
19:13:44.663 -> Saved PC:0x40380000
19:13:44.663 -> SPIWP:0xee
19:13:44.663 -> mode:DIO, clock div:1
19:13:44.663 -> load:0x3fcd5810,len:0x438
19:13:44.663 -> load:0x403cc710,len:0x90c
19:13:44.663 -> load:0x403ce710,len:0x25f4
19:13:44.663 -> entry 0x403cc710
19:13:44.835 -> Starting ESP32 BLE

NO advertisement is this normal?

ckcr4lyf, 2024-01-06:
Is the serial output? There should be some lines printed if you used the exact program that is in this repository.

Did you use Arduino IDE or PlatformIO?

silverdragon830, 2024-01-06:
yes this is the serial output, using platofrmio, and usedd the exact program

silverdragon830, 2024-01-06:
i just tried again with arduino ide and it works flawless thank you so much

btw admin do you have more projects like this very fun for my 5year old son

ckcr4lyf, 2024-01-08:
Glad you got it working. I've not other projects like this, if I do something I'll let you know.

closed 2024-01-08 by ckcr4lyf