ckcr4lyf/EvilAppleJuice-ESP32 #26: Vision pro

sionk99, 2024-01-17:
[image]
能否更新一下这个代码

ckcr4lyf, 2024-01-18:
Sorry, the photo hasn't seem to be uploaded correctly. Can you try again?

Are you asking for support on the Vision Pro?

sionk99, 2024-01-18:
[image]

sionk99, 2024-01-18:
能否更新一下这个

ckcr4lyf, 2024-01-18:
Ah, understood. Let me try and find out what the code for it is. Is there any other project you know using it?

ckcr4lyf, 2024-01-18:
@sionk99 do you have an iPhone? I do not have one so I cannot test it, but maybe you can try changing the 14th byte in the payload of `SHORT_DEVICES` at https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/blob/78984d5f1ff1ca0444caf02ff1cfd9b300c60f0d/src/devices.hpp#L50

It seems this is the changing byte between different devices, so maybe it is possible Vision Pro uses one of these.

[image]

ckcr4lyf, 2024-01-18:
Maybe slowly try all of them from 0x00-0xFF... not a great solution but for the time being that is all I can offer. I will try and work on the problem if I get access to an iPhone.

sionk99, 2024-01-18:
https://github.com/simondankelmann/Bluetooth-LE-Spam        这个项目的，安卓app

ckcr4lyf, 2024-01-18:
@sionk99 I've made a PR at https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/pull/27 . 

I've not tested this yet so I am not merging it, but you can try and pull that branch and see if it works!

ckcr4lyf, 2024-01-19:
Added via #27 .

closed 2024-01-19 by ckcr4lyf