ckcr4lyf/EvilAppleJuice-ESP32 #39: compilation erroe

laotai090, 2024-06-23:
if (device_choice == 0){
    int index = random(17);
    oAdvertisementData.addData(std::string((char*)DEVICES[index], 31));
  } else {
    int index = random(13);
    oAdvertisementData.addData(std::string((char*)SHORT_DEVICES[index], 23));
  }

Compilation error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String

ckcr4lyf, 2024-06-23:
@laotai090 can you share a bit more about the platform you're using? E.g. VS Code / PlatformIO / VS Code?

Maybe some stuff here will help: https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/issues/12#issuecomment-1794585749

laotai090, 2024-06-24:
arduino ide and vscode both cannot compilation, it say std::string((char*)DEVICES[index], 31 have error

ckcr4lyf, 2024-06-24:
If you can share your platformio ini file, and a more detailed message (e.g. with full trace) or screenshot of the error, it would be helpful.

PogusTheWhisper, 2024-06-30:
D:\0.Users\pogus-asus\Downloads\EvilAppleJuice-ESP32-master\EvilAppleJuice-ESP32-master\src\EvilAppleJuice-ESP32-INO\EvilAppleJuice-ESP32-INO.ino: In function 'void loop()':
D:\0.Users\pogus-asus\Downloads\EvilAppleJuice-ESP32-master\EvilAppleJuice-ESP32-master\src\EvilAppleJuice-ESP32-INO\EvilAppleJuice-ESP32-INO.ino:128:37: error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
  128 |     oAdvertisementData.addData(std::string((char*)DEVICES[index], 31));
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                     |
      |                                     std::string {aka std::__cxx11::basic_string<char>}
In file included from C:\Users\pogus-asus\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLEServer.h:22,
                 from C:\Users\pogus-asus\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLEDevice.h:21,
                 from D:\0.Users\pogus-asus\Downloads\EvilAppleJuice-ESP32-master\EvilAppleJuice-ESP32-master\src\EvilAppleJuice-ESP32-INO\EvilAppleJuice-ESP32-INO.ino:5:
C:\Users\pogus-asus\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLEAdvertising.h:36:23: note:   initializing argument 1 of 'void BLEAdvertisementData::addData(String)'
   36 |   void addData(String data);  // Add data to the payload.
      |                ~~~~~~~^~~~
D:\0.Users\pogus-asus\Downloads\EvilAppleJuice-ESP32-master\EvilAppleJuice-ESP32-master\src\EvilAppleJuice-ESP32-INO\EvilAppleJuice-ESP32-INO.ino:131:37: error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'
  131 |     oAdvertisementData.addData(std::string((char*)SHORT_DEVICES[index], 23));
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                     |
      |                                     std::string {aka std::__cxx11::basic_string<char>}
C:\Users\pogus-asus\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\BLE\src/BLEAdvertising.h:36:23: note:   initializing argument 1 of 'void BLEAdvertisementData::addData(String)'
   36 |   void addData(String data);  // Add data to the payload.
      |                ~~~~~~~^~~~

exit status 1

Compilation error: cannot convert 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'String'

ckcr4lyf, 2024-07-02:
Can you share the version of PlatformIO you're using, the chip you're trying to flash, and any other PlatformIO specific config? (e.g. version of ESP32-Arduino etc.)

ckcr4lyf, 2024-07-07:
I think I've found an issue in the Arduino version. I was only using PlatformIO so it slipped past, sorry.

ckcr4lyf, 2024-07-07:
@laotai090 @PogusTheWhisper i've pushed a fix for the Arduino version, can you guys try it out and let me know if it works?

Tested via:

```
Used library Version Path
BLE          2.0.0 

Used platform Version Path
esp32:esp32   3.0.2 
```

ckcr4lyf, 2024-07-08:
The new conditional compilation should fix. If you still have issues please reopen.

PR #41 (2024-07-08): Conditional Compilation for ESP32-Arduino v2/v3 breaking changes
https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html#ble

The `std::string` -> `String` is was causing some problems such as #39 .

This should fix it, so no matter which version you're on, it should work.

Closes #39
closed 2024-07-08 by ckcr4lyf