ckcr4lyf/EvilAppleJuice-ESP32 #42: Optional

theguythatdidsomething, 2024-08-05:
Hey there what's up i am interested in only spamming one product without crashing the iPhone is it possible. 

Thanks for your time

ckcr4lyf, 2024-08-05:
AFAIK this should no longer crash an iPhone (DoS / reboot), that was patched long ago. It will just spam pop-ups.

You can modify the randomness to instead just pick a specific product and spam that, though I think the effectiveness won't be as good.

ckcr4lyf, 2024-08-05:
https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/blob/6ef1aa131f5fa7fd36114a584dc9fee55f96da6e/src/main.cpp#L72-L91

you might find this part interesting

ckcr4lyf, 2024-08-05:
Which device do you have in mind? I can help you patch it for that. (You can view the list here: https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/blob/6ef1aa131f5fa7fd36114a584dc9fee55f96da6e/src/devices.hpp)

theguythatdidsomething, 2024-08-05:
Vicion pro and hopefully when the esp32 is connected to power it only send the pairing once every 10 thanks

theguythatdidsomething, 2024-08-05:
Basically want I need is a code easy to edit the product to advertise and the seconds to not be bothering you every time I try something new @ckcr4lyf . 

Ps thanks so much for you're help people like you is what the world needs nice people willing to help people strangers no matter what maybe i took it to far there but thanks

theguythatdidsomething, 2024-08-05:
Last thing my esp32 is arriving the 12 of this month so take your time

ckcr4lyf, 2024-08-05:
I do not use discord for such purposes since it is a walled off platform.

I'll help you patch it for the Vision Pro in a bit and explain it as well, so that hopefully you can do the next one yourself.

theguythatdidsomething, 2024-08-05:
When do you  you'll be ready no pressure just curious

theguythatdidsomething, 2024-08-06:
I think you should make this a part on the main page how to select device

theguythatdidsomething, 2024-08-07:
Tried studying you're code to true and make the custom spam  let's Call it but it didn't go well

theguythatdidsomething, 2024-08-07:
@ckcr4lyf  Any luck in patching the code

ckcr4lyf, 2024-08-07:
Whenever it is done you will see an update here.

theguythatdidsomething, 2024-08-17:
any updates have you had luck

ckcr4lyf, 2024-08-17:
How to spam a single device:

1. Determine the device from https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/blob/master/src/devices.hpp
2. If it is in `DEVICES`, change `random(2)` to `0` on this line: https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/blob/6ef1aa131f5fa7fd36114a584dc9fee55f96da6e/src/main.cpp#L72 , otherwise change it to `1`
3. Count the number of the devices in `DEVICES` or `SHORT_DEVICES` , starting from zero.
4. Change either https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/blob/6ef1aa131f5fa7fd36114a584dc9fee55f96da6e/src/main.cpp#L74 or https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/blob/6ef1aa131f5fa7fd36114a584dc9fee55f96da6e/src/main.cpp#L83 line to the number, depending on which type of device it is.

Example for "Airpods Max"

It is the 3rd device in `DEVICES`.

So we will change https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/blob/6ef1aa131f5fa7fd36114a584dc9fee55f96da6e/src/main.cpp#L72 to ` = 0;` at the end , and https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/blob/6ef1aa131f5fa7fd36114a584dc9fee55f96da6e/src/main.cpp#L74 to `= 2;` at the end.

ckcr4lyf, 2024-08-17:
That's about as much as I'm willing to help, other than that I'd suggest learning some C++ if you're still confused.

closed 2024-08-17 by ckcr4lyf

theguythatdidsomething, 2024-08-17:
#include <Arduino.h>
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <esp_arduino_version.h>

// Bluetooth maximum transmit power
#if defined(CONFIG_IDF_TARGET_ESP32C3) || defined(CONFIG_IDF_TARGET_ESP32C2) || defined(CONFIG_IDF_TARGET_ESP32S3)
#define MAX_TX_POWER ESP_PWR_LVL_P21  // ESP32C3 ESP32C2 ESP32S3
#elif defined(CONFIG_IDF_TARGET_ESP32H2) || defined(CONFIG_IDF_TARGET_ESP32C6)
#define MAX_TX_POWER ESP_PWR_LVL_P20  // ESP32H2 ESP32C6
#else
#define MAX_TX_POWER ESP_PWR_LVL_P9   // Default
#endif

BLEAdvertising *pAdvertising;  // global variable
uint32_t delayMilliseconds = 1000;

// AirPods Max specific advertisement data (example, replace with actual data)
const uint8_t AIRPODS_MAX_DATA[] = {
    0x02, 0x01, 0x06, 0x1A, 0xFF, 0x4C, 0x00, 0x02, 0x15, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};

// Function to generate a random MAC address
void generateRandomMAC(esp_bd_addr_t &addr) {
    for (int i = 0; i < 6; i++) {
        addr[i] = random(256);
    }
    addr[0] |= 0x02;  // Set the Local MAC address type (LAP) to Random
}

void setup() {
    Serial.begin(115200);
    Serial.println("Starting ESP32 BLE");

    // This is specific to the AirM2M ESP32 board
    pinMode(12, OUTPUT);
    pinMode(13, OUTPUT);

    BLEDevice::init("AirPods Max");

    // Increase the BLE Power to 21dBm (MAX)
    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, MAX_TX_POWER);

    // Create the BLE Server
    BLEServer *pServer = BLEDevice::createServer();
    pAdvertising = pServer->getAdvertising();

    Serial.println("BLE Setup Complete");
}

void loop() {
    // Turn lights on during "busy" part
    digitalWrite(12, HIGH);
    digitalWrite(13, HIGH);

    // Generate a random MAC address
    esp_bd_addr_t randomAddr;
    generateRandomMAC(randomAddr);

    BLEAdvertisementData oAdvertisementData = BLEAdvertisementData();

    // Use AirPods Max specific advertisement data
    #ifdef ESP_ARDUINO_VERSION_MAJOR
        #if ESP_ARDUINO_VERSION >= ESP_ARDUINO_VERSION_VAL(3, 0, 0)
            oAdvertisementData.addData(String((char*)AIRPODS_MAX_DATA, sizeof(AIRPODS_MAX_DATA)));
        #else
            oAdvertisementData.addData(std::string((char*)AIRPODS_MAX_DATA, sizeof(AIRPODS_MAX_DATA)));
        #endif
    #endif

    // Set advertisement type for AirPods Max
    pAdvertising->setAdvertisementType(ADV_TYPE_SCAN_IND);

    // Set the advertisement data
    pAdvertising->setAdvertisementData(oAdvertisementData);

    // Set the device address
    pAdvertising->setDeviceAddress(randomAddr, BLE_ADDR_TYPE_RANDOM);

    // Start advertising
    Serial.println("Starting Advertisement...");
    pAdvertising->start();

    // Add a delay to ensure advertisement starts
    delay(2000);

    Serial.println("Stopping Advertisement...");
    // Turn lights off while "sleeping"
    digitalWrite(12, LOW);
    digitalWrite(13, LOW);
    delay(delayMilliseconds); // delay for delayMilliseconds ms
    pAdvertising->stop();

    // Random signal strength increases the difficulty of tracking the signal
    int rand_val = random(100);  // Generate a random number between 0 and 99
    if (rand_val < 70) {  // 70% probability
        esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, MAX_TX_POWER);
    } else if (rand_val < 85) {  // 15% probability
        esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, (esp_power_level_t)(MAX_TX_POWER - 1));
    } else if (rand_val < 95) {  // 10% probability
        esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, (esp_power_level_t)(MAX_TX_POWER - 2));
    } else if (rand_val < 99) {  // 4% probability
        esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, (esp_power_level_t)(MAX_TX_POWER - 3));
    } else {  // 1% probability
        esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, (esp_power_level_t)(MAX_TX_POWER - 4));
    }
}
 not workig did as you said

ckcr4lyf, 2024-08-17:
If you'd like help regarding programming, please ask on Stack Overflow or similar. I will only support bugs with the main program in this repo.

theguythatdidsomething, 2024-08-17:
ok sorry for bothering thanks for ur time

theguythatdidsomething, 2024-08-17:
then can you give me the airpods max code

PR #45 (2024-08-17): Some guidance for single device
Closes #42