ckcr4lyf/EvilAppleJuice-ESP32 #46: Range Problem

biscuitaep, 2024-11-16:
I have NodeMCU ESP32S V1.1 Module (ESP WROOM32) and the advertising range is less than 40cm's how can I fix that?

iOS Version: 18.1

ckcr4lyf, 2024-11-16:
Hi buiscuitaep,

I do not have that exact model, so I'm not sure I can be of too much help. It seems like the max power level for your chip might be  `ESP_PWR_LVL_N12` .

Can you change this line: https://github.com/ckcr4lyf/EvilAppleJuice-ESP32/blob/28d6be64f0417d51cad37c4b67ac01619f8fae0f/src/main.cpp#L19 to `ESP_PWR_LVL_N12` , and see if that helps?

biscuitaep, 2024-11-16:
> Hi buiscuitaep,
>
>
>
> I do not have that exact model, so I'm not sure I can be of too much help. It seems like the max power level for your chip might be  `ESP_PWR_LVL_N12` .
>
>
>
> Can you change this [...]

Now it isnt working, isnt P9 best value?

ckcr4lyf, 2024-11-16:
I'm not too sure about your specific device.

I can recommend purchasing a ESP32C3, which is what I've tested with and performs well in my experience.

biscuitaep, 2024-11-16:
Ok Ty, Also when I select long range devices (short like appletv) nothing happens why do you know?

ckcr4lyf, 2024-11-16:
\>  select long range devices (short like appletv)

Do you mean AppleTV is an example of the device where "nothing happens", or AppleTV is a "short device"?

ckcr4lyf, 2025-03-19:
Closing due to lack of information

closed 2025-03-19 by ckcr4lyf