ckcr4lyf/EvilAppleJuice-ESP32 #48: ESP32C3 Range is 5cm after 5cm only apple tv works

codeplxor, 2025-03-05:
ESP32C3 Range is 5cm after 5cm only apple tv works
nothing modified on script

ckcr4lyf, 2025-03-05:
Which manufacturer's ESP32-C3 are you using? What type of iOS devices are in the immediate (say, upto 5 meters) vicinity?

> only apple tv works

Do you mean that's the only pop-up?

codeplxor, 2025-03-05:
Im using the HW-466AB ESP32-C3 Supermini
On older devices, at long range only Apple TV stuff works
But on my iPhone XS iOS 18 At very close range only speaker things work at less than 5 cm.

ckcr4lyf, 2025-03-06:
> Im using the HW-466AB ESP32-C3 Supermini

Honestly, I think it's the board that's the problem. The Supermini has a much smaller antenna than something like the ESP32-C3 CORE.

Regardless, I will try to retest my ESP32C3 & ESP32C6 in case it may be due to an iOS update.

codeplxor, 2025-03-06:
Ty gonna buy the regular soon